OpenRoberta/robertalab #7: EV3 debugging

Malidox, 2015-12-03:
1. The user drags a block "start logging" from the control category to the NEPO program. From this point, the robot will output sensor values regularly. This blockly block will not 'block' program execution.
2. In this block the user can choose if the robot should log to the server or to the ev3 screen or both.
3. The user can only log sensors/ sensor modes which are used in the nepo program.
4. The user can see the values in realtime on the top right of the blockly window in the web application. The server updates those values regularly.
5. The user can see the values in realtime on the robot screen. Because the space is so limited, the lines scroll down automatically if more text arrives (similar to command line output).
6. The user drags a block "stop logging" from the control category to the NEPO program. From this point, the robot will no longer output sensor values any more.
7. The user wants to see if specific parts of the programs are reached during runtime. The user drags a "system out" block from the control category to the NEPO program. The user types a string in (maybe limit to 16 characters) and specifies, if the program should output this text to the Open Roberta server or to the robot screen or both.
8. The program reaches the block during runtime and will output the text.

ensonic, 2015-12-04:
I though to be pragmatic and add 'enter'+'leave' hooks to each block. That is when generating code we generate: "enter(<block-id>);<block-code>;leave(<block-id>);" Now the code in enter/leave can send a message to the server (websocket) and that can only contain the block id or also sensor data. Using enter/leave avoid the extra thread and the arm9 is notoriously bad at context switching

The idea to use a logging block is okay, but i wonder if in most case people just want to enable or disable it globally. Also I'd be worried if you add blocks to enable disable logging, I can imagine people filing feature requests to ask for filters (what to log) and then the next step is to have your own debug.printf() block anyway :/

If we unconditionally log the sensor feed to the server we can also consider things like small animations in the blocks, e.g. the color sensor could show the color/brightness in an icon as the program runs and in a motor block we can have a spinnign animation when the motor runs that is proportional to the speed.

bjost2s, 2016-04-06:
In 1.5 it is supposed to be extended. The first version, enable sensor logging on the robot is done for lejos.

closed 2019-02-01 by VinArt