OpenRoberta/robertalab #8: Data exchange between EV3 robot and server

Malidox, 2015-12-03:
Technical description about logging/ debugging NEPO programs:
1. The NEPO program will create a websocket to send data to the server, currently we do not send anything back but this can change in the future
2. The NEPO program will get the last used token and the server address from the menu via a properties file
3. We only log sensors and their sensor modes which are actually used in the NEPO program, the code generation has to be changed therefor by Kostadin
4. The logging calls the same methods from hal as the nepo program but from a different thread, hal methods are now synchronized and can delay the execution of the NEPO program by a short amount of time
5. Logging interval is ones per second, may be increased if this is too demanding on the hardware
6. Beate will add new control blocks for switching the logging on and off
7. We want to use a highlight block which will send a block id to the server, if the websocket is not yet instantiated because logging is turned off, this will be done on demand
8. We also add blocks for activating/ deactivating logging to the EV3 display using the LCDOutputStream class from lejos. This screen logging is separate from the remote logging

ensonic, 2015-12-04:
If we do screen logging, I would not use a LCDOutputStream since that will overwrite the screen in a uncontrolled way. I think for that case we want to find a way to e.g. show the status in a region of the screen so that the user can still do other screen output. This could be to format the sensor/motor status into two lines of text that the user could output e.g. on first and last line of the screen.

bjost2s, 2016-04-06:
How can we test this?

dpyka, 2016-04-06:
1) Currently, we check if the file /home/roberta/lib/Java-WebSocket.jar exists. Log in via SSH and rename it to something else to test it.
2) If the websocket fails to connect, we use REST as fallback solution, but i do not know any environment where we can test this :-)

closed 2019-02-01 by VinArt