OpenRoberta/robertalab #10: Please don't include (large) binary files in git

dlech, 2015-12-04:
Cloning this repository today is already 350MB. I can see that there are binary files such as .jar, .deb and .msi files included. When you clone git, you get every copy of this file ever committed, which is probably why the download is so big.

At the very least, you should stop committing these files. If you don't mind forcing everyone to clone a new copy of the repository, you can rewrite history using `git filter-branch` to remove the large files. Or, if you think it is important to have these files in git, have a look at <https://git-lfs.github.com/>.

dpyka, 2016-01-04:
In the develop branch, i put the .msi and .deb on gitignore. I also removed the jars from the runtime project, which were actually never used :-)
What i can see, there is only ev3control.jar left, which is quite handy for capturing the ev3 screen to make pictures for our wiki.

ensonic, 2016-01-05:
The ev3control.jar is lejos specific. Can we just link this from a devel-wiki page?

closed 2016-02-10 by dpyka