OpenRoberta/robertalab #13: suggestions: set default branch on github to "develop"

dlech, 2015-12-05:
In the settings for each repository, github allows you to select the default branch that is displayed when anyone visits the site. If you have master as the default, then people will do like I just did and send a pull request to master instead of develop. :grin:

bjost2s, 2015-12-05:
We have had that in the old repository and switched to master as default. We have people (none developers) that clone the repository to run the server locally. It is hard to explain for them how to switch to master.

dlech, 2015-12-05:
That makes sense. Another option is to tag releases. GitHub automatically provides .zip and .tar.gz files for each tag. You could point people that want a known working release to download from there instead.

Or you should add a note to the readme in master (because that is the one people are going to see) that explains your workflow.

ensonic, 2015-12-11:
So actually at least until the next release we're doing new changes against develop.

closed 2016-12-21 by bjost2s