OpenRoberta/robertalab #19: Allow '_' in program names.

ensonic, 2015-12-14:
As requested by the user, we should allow more chars for the program name. Right now we use the program name also for the main class and hence enforce the CamelCase style typical for java.

I think we should at least allow also the '_' and maybe even other chars and just canonicalize the name when deriving the class name.

rbudde, 2016-02-23:
The server accepts all valid Java identifier (for example "__üäö$€"). There is a (somehow redundant) javascript check at client side. I'll changed the client-side pattern there to
^[a-zA-Z_öäüÖÄÜß$€][a-zA-Z0-9_öäüÖÄÜß$€]*$
This cures the problem. But this is only a temporary solution for 1.4, as this kind of redundancy is bad.
Commit is 5b7380a

rbudde, 2016-02-23:
is the fix ok for you? Then please close.

closed 2016-02-23 by ensonic