OpenRoberta/robertalab #22: Only 64Bit?

thski, 2015-12-23:
When I started via ora.sh, the server didn't start because the script complained that my java version is only 32bit. I deleted the lines 

    case "$javaversion" in
        *64-Bit*) : ;;
        *)        checkJava="${checkJava}  java is resolved to a 32 bit version. The server needs a 64 bit jdk.\n" ;;
    esac

and it seemed to work even on a 32bit machine.

bjost2s, 2015-12-23:
That's right. We have already changed that in our develop branch. So for the next release this test will only check whether there is a jdk 1.7 installed. Thanks!

yajo10, 2016-05-12:
You have to do this in your main open-roberta folder: `$ sudo nano ora.sh` and find the line `_checkJava` and change it to `# _checkJava` like in this [https://mp-devel.iais.fraunhofer.de/wiki/display/ORInfo/Open+Roberta+Standalone](url) descriped.

closed 2017-02-25 by bjost2s