OpenRoberta/robertalab #24: Allow choice of programming language ("show code")

bittner, 2016-01-01:
In [Open Roberta Lab](http://lab.open-roberta.org/) when the button `<>` (show code) or the menu item "show code" is pressed a page of generated Java code is shown (58 lines of code for an empty program).

It would be nice if a user could specify her/his preferred programming language, so that instead of Java code (by default) some other language (JavaScript, Python, Lua, ...) would be shown. This is supported by Blocky [according to its documentation](https://developers.google.com/blockly/installation/code-generators).

For educational purposes a less verbose language than Java may be advisable as the default.

ensonic, 2016-01-02:
Right now, the openroberta lab is generating java code, since it targets lejos which is based on java. I wrote an openrobertalab adapter for ev3dev (already packaged as 'openrobertalab'). This will use python code. While blockly also supports different languages, those won't be supported on the robot for now. On ev3dev we can potentially support additional languages, but this is not a priority right now.

To give a little more background - the openroberta lab is not using the code-generators blockly provides. It generates the code on the server side and does some additional checking.

bittner, 2016-01-03:
I understand. Interesting though that Java is used for an education project. For programmers this may be interesting, I agree, but for newbie programmers (kids) the verbosity of Java kills all connections to the logic constructed using blocks.

On Code.org with the new [Star Wars game](https://studio.code.org/s/starwars/), for example, using blocks and JavaScript it comes so natural for kids to try writing IntelliSense-supported code after doing coding just with blocks. One line of code roughly matches one block used earlier. This is simply not possible with Java, because it's a verbose language by design.

As I said, not a big problem for a programmer, but a potential show stopper for a learning child. I feel that especially in education less verbose languages are more adequate. As you become more used to programming over time optionally switching to a language like Java could make a lot of sense.

ensonic, 2016-01-03:
What kind of resolution do you hope for with regard to this ticket?

Speaking of your example - the star wars game - when targeting a simulation in the browser java script is a logical choice. When targeting real devices, we face much more diverse environments.

Regarding the programming languages. As I said lejos as a ev3 firmware targets java only. When using the ev3dev.org firmware we are using python which should be more what you are looking for. While ev3dev.org also supports javascript (via nodejs), it is not a future proof choise. The ev3 is running a very old arm9 processor which can only run node.js up to 0.10, but newer versions.

OpenRoberta will also target other robotic systems in the future and for some of them there might be even less of a choise in language (e.g. only c for arduino based systems). Comming up with a smooth transitions from blocks to real code is indeed a challenge and we hope to improve this in future version of the software.

Maybe you can tell us more in which context you like to use openroberta with ev3s?

bittner, 2016-01-03:
In fact I would love to see Python as a default. JavaScript is just what the guys at Code.org most obviously seem to prefer.

I would like to use Open Roberta (*) as an easy-to-use Web IDE for programming (LEGO) robots. I have Linux PCs for my kids, but there is no entry-level LEGO software to program the EV3 brick on something other than Windows or OS X. I'd love to see them start to program the LEGO brick with Blockly blocks, and then seamlessly switch to coding Python, as (successfully proved) with the Star Wars example.

Anyone could start (learn) programming little robots like this. And isn't the Open Roberta project about "anyone should be able to learn programming robots"?

------
(*) The open source project is not allowed to use that name outside of Fraunhofer-Institut according to the footer at http://dev.open-roberta.org, correct? What about "Roberta IDE" or "Roberta Libre", a Cuba libre flavor of the IDE?  :smirk:

ensonic, 2016-01-04:
Okay, thenwe're super aligned :-) My work on the openrobertalab package on ev3dev is exactly to make this the default. You can already use it, if you run the latest ev3dev images and install the "openrobertalab" package. The lab.openroberta.org server already contain the needed code. When connected to an ev3 running ev3dev, the code should be shown as python. To be honest I am not totally sure if this is already contained in the version running on the server. So if you still see java code, you'll get python once we update the server release.

bjost2s, 2016-04-06:
for downloading the code the gui produces currently a .java file which is not really good for python code. Is there a way, that the gui gets the information, which firmware the ev3 uses?

ensonic, 2016-04-07:
@bjost2s the gui creates java if an ev3 running lejos is connected and python if the ev3 runs ev3dev. Before you connect a robot, it defaults to java. So if no robot is connected we could have a language selector (also nice for testing).

ensonic, 2017-01-19:
@bjost2s - we still need a solution for selecting the language if no robot is connected. If we split ev3 and lejos we have two ev3 variants and then this is implicitly fixed.

ensonic, 2017-01-19:
instead of #363, see #308

DJuego, 2017-02-10:
I make public my support for this feature. I have the honor of teaching with ev3dev: Open Roberta Lab connector  and  ev3dev-lang-python. I miss this feature a lot. :-/

bjost2s, 2017-02-27:
I think with splitting lejos and ev3dev in two different plugins, we will have the possibility to add more and different features (blocks) to ev3dev. But with this, we could not support the code view for python and java at the same time.

commit baf45ac7 (2016-04-10): fix #24 with correct file extension
closed 2019-02-01 by VinArt