OpenRoberta/robertalab #25: USB Connection - avoid reenter password after reconnection / Improvement-suggestion for an easier local installation

thski, 2016-01-10:
I have managed to install the server software on my linux machine and use the same machine for programming with the browser client. So far so good.
What I could not find previously is any information if I can use a usb connection. I tried it and it works very well. By this one is not dependent on any network connection (I don't want my 10 year old son to be connected to the internet all day). At school this may be useful, too! Perhaps this also could be mentioned in some howto. Perhaps a downloadable zip-file with integrated server-client-environment for local installation for using it via usb-connection would be great and more user-friendly.

But I realized that if I unplug the ev3 for test purposes and reconnect it later in order to improve the program, I have to restart the connection process, that means I have to reenter a new password. This is somehow annoying, so perhaps it is possible to avoid this?

thski, 2016-01-10:
Just found this project: https://github.com/OpenRoberta/robertalab-chromeapp
Haven't tried it yet, but knowing this would have made things easier right from scratch...

dpyka, 2016-01-11:
Hello, thank you for your feedback. Connecting the brick to 10.0.1.10:1999 via the menu and usb cable is not recommended, because of this reconnecting issue.
You may want to have a look at our Chrome App. There you can select another host, if you are not using lab.open-roberta.org (See the screenshot below). The screenshot shows the german version but english is also available! The Chrome App wil either connect to the official server, or to the host you specifiy in the advanced options. So this would be the solution you are looking for.

[image]

Here you can find our wiki page for additional information about the Chrome App:
https://mp-devel.iais.fraunhofer.de/wiki/display/ORInfo/Open+Roberta+Chrome+App

Have fun!

ensonic, 2017-01-23:
@thski did you use lejos or ev3dev. With ev3dev you don't need to reconnect anymore (see OpenRoberta/robertalab-ev3dev#31).

thski, 2017-01-24:
Great news!
I did use ev3dev. I will try it again. Thank you!

ThSki

ensonic, 2017-01-24:
@thski FYI, I#ll release an update in the next days. It require updating this package on the ev3 then,

closed 2019-02-01 by VinArt