OpenRoberta/robertalab #26: Compiler workflow error not handled correctly in push protocol

dpyka, 2016-01-14:
In case of compiler workflow error (code cannot be compiled), the brick still tries to download and run the program, which does not exist (the ev3 tries to run the directory because of the missing file name :-))

I think it is not a big thing, but i need to check if this is an error on server side or on the brick.

11:41:16.561 [main] INFO de.fhg.iais.roberta.util.Util - properties from classpath. Using the resource: openRoberta.properties
11:41:16.576 [main] INFO d.f.iais.roberta.util.VersionChecker - version of OpenRobertaRuntime is 1.3.2-SNAPSHOT and considered ok
11:41:16.576 [main] INFO d.f.iais.roberta.util.VersionChecker - version of OpenRobertaServer is 1.3.2-SNAPSHOT and considered ok
11:41:16.576 [main] INFO d.f.iais.roberta.util.VersionChecker - version of OpenRobertaShared is 1.3.2-SNAPSHOT and considered ok
11:41:16.592 [main] INFO org.eclipse.jetty.util.log - Logging initialized @484ms
11:41:16.701 [main] INFO org.eclipse.jetty.server.Server - jetty-9.2.5.v20141112
Jan 14, 2016 11:41:17 AM com.sun.jersey.api.core.PackagesResourceConfig init
Information: Scanning for root resource and provider classes in the packages:
  de.fhg.iais.roberta.javaServer.restServices.all
  de.fhg.iais.roberta.javaServer.restServices.ev3
  de.fhg.iais.roberta.javaServer.provider
Jan 14, 2016 11:41:17 AM com.sun.jersey.api.core.ScanningResourceConfig logClasses
Information: Root resource classes found:
  class de.fhg.iais.roberta.javaServer.restServices.ev3.Ev3Command
  class de.fhg.iais.roberta.javaServer.restServices.all.RestExample
  class de.fhg.iais.roberta.javaServer.restServices.all.Alive
  class de.fhg.iais.roberta.javaServer.restServices.all.ClientToolbox
  class de.fhg.iais.roberta.javaServer.restServices.all.ClientUser
  class de.fhg.iais.roberta.javaServer.restServices.ev3.Ev3Update
  class de.fhg.iais.roberta.javaServer.restServices.ev3.Ev3DownloadJar
  class de.fhg.iais.roberta.javaServer.restServices.all.ClientProgram
  class de.fhg.iais.roberta.javaServer.restServices.all.ClientPing
  class de.fhg.iais.roberta.javaServer.restServices.all.ClientConfiguration
  class de.fhg.iais.roberta.javaServer.restServices.all.ClientAdmin
Jan 14, 2016 11:41:17 AM com.sun.jersey.api.core.ScanningResourceConfig logClasses
Information: Provider classes found:
  class de.fhg.iais.roberta.javaServer.provider.JAXBContextProvider
  class de.fhg.iais.roberta.javaServer.provider.JAXBUnmarshallerProvider
  class de.fhg.iais.roberta.javaServer.provider.JAXBMarshallerProvider
  class de.fhg.iais.roberta.javaServer.provider.OraDataProvider
Jan 14, 2016 11:41:17 AM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
Information: Registering de.fhg.iais.roberta.javaServer.restServices.all.ClientAdmin as a root resource class
Jan 14, 2016 11:41:17 AM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
Information: Registering de.fhg.iais.roberta.javaServer.restServices.all.ClientConfiguration as a root resource class
Jan 14, 2016 11:41:17 AM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
Information: Registering de.fhg.iais.roberta.javaServer.restServices.all.ClientToolbox as a root resource class
Jan 14, 2016 11:41:17 AM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
Information: Registering de.fhg.iais.roberta.javaServer.restServices.all.ClientProgram as a root resource class
Jan 14, 2016 11:41:17 AM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
Information: Registering de.fhg.iais.roberta.javaServer.restServices.all.ClientUser as a root resource class
Jan 14, 2016 11:41:17 AM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
Information: Registering de.fhg.iais.roberta.javaServer.restServices.ev3.Ev3DownloadJar as a root resource class
Jan 14, 2016 11:41:17 AM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
Information: Registering de.fhg.iais.roberta.javaServer.restServices.ev3.Ev3Command as a root resource class
Jan 14, 2016 11:41:17 AM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
Information: Registering de.fhg.iais.roberta.javaServer.restServices.all.RestExample as a root resource class
Jan 14, 2016 11:41:17 AM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory register
Information: Registering de.fhg.iais.roberta.javaServer.restServices.all.ClientPing as a root resource class
Jan 14, 2016 11:41:17 AM com.sun.jersey.server.impl.application.WebApplicationImpl _initiate
Information: Initiating Jersey application, version 'Jersey: 1.18.2 10/26/2014 03:51 PM'
Jan 14, 2016 11:41:17 AM com.sun.jersey.guice.spi.container.GuiceComponentProviderFactory getComponentProvider
Information: Binding de.fhg.iais.roberta.javaServer.provider.OraDataProvider to GuiceInjectedComponentProvider
11:41:17.587 [main] INFO o.h.annotations.common.Version - HCANN000001: Hibernate Commons Annotations {4.0.2.Final}
11:41:17.587 [main] INFO org.hibernate.Version - HHH000412: Hibernate Core {4.2.7.Final}
11:41:17.587 [main] INFO org.hibernate.cfg.Environment - HHH000206: hibernate.properties not found
11:41:17.587 [main] INFO org.hibernate.cfg.Environment - HHH000021: Bytecode provider name : javassist
11:41:17.619 [main] INFO org.hibernate.cfg.Configuration - HHH000043: Configuring from resource: hibernate-cfg.xml
11:41:17.619 [main] INFO org.hibernate.cfg.Configuration - HHH000040: Configuration resource: hibernate-cfg.xml
11:41:17.682 [main] INFO org.hibernate.cfg.Configuration - HHH000041: Configured SessionFactory: null
11:41:17.729 [main] INFO o.h.s.j.c.i.ConnectionProviderInitiator - HHH000130: Instantiating explicit connection provider: org.hibernate.service.jdbc.connections.internal.C3P0ConnectionProvider
11:41:17.729 [main] INFO o.h.s.j.c.i.C3P0ConnectionProvider - HHH010002: C3P0 using driver: org.hsqldb.jdbcDriver at URL: jdbc:hsqldb:file:db/openroberta-db
11:41:17.729 [main] INFO o.h.s.j.c.i.C3P0ConnectionProvider - HHH000046: Connection properties: {user=orA, password=****, autocommit=false}
11:41:17.729 [main] INFO o.h.s.j.c.i.C3P0ConnectionProvider - HHH000006: Autocommit mode: false
Jan 14, 2016 11:41:17 AM com.mchange.v2.log.MLog <clinit>
Information: MLog clients using java 1.4+ standard logging.
Jan 14, 2016 11:41:17 AM com.mchange.v2.c3p0.C3P0Registry banner
[... 269 lines cut ...]
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1652) [jetty-servlet-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:585) [jetty-servlet-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:221) [jetty-server-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1125) [jetty-server-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:515) [jetty-servlet-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185) [jetty-server-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1059) [jetty-server-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141) [jetty-server-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.server.handler.HandlerList.handle(HandlerList.java:52) [jetty-server-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:97) [jetty-server-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.server.Server.handle(Server.java:497) [jetty-server-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:310) [jetty-server-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:248) [jetty-server-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540) [jetty-io-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:620) [jetty-util-9.2.5.v20141112.jar:9.2.5.v20141112]
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:540) [jetty-util-9.2.5.v20141112.jar:9.2.5.v20141112]
[... cut ...]
11:42:08.910 [qtp1902465477-14] ERROR d.f.i.r.j.r.ev3.Ev3DownloadJar - jar could not be uploaded to robot: jar to upload to robot not found
11:42:10.892 [qtp1902465477-12] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 28 msec: run program 'NEPOprog' with configuration 'EV3basis'
11:42:10.892 [qtp1902465477-12] INFO d.fhg.iais.roberta.util.ClientLogger - 1 log entry written
11:49:35.966 [qtp1902465477-16] INFO d.f.i.r.j.r.all.ClientPing - /ping [count:100]
11:57:56.240 [qtp1902465477-12] INFO d.f.i.r.j.r.all.ClientPing - /ping [count:200]

dpyka, 2016-01-15:
Whole program is no longer printed to console.
+Some small fixes to existing logging.

ensonic, 2016-01-29:
Is this now fixed?

closed 2016-02-02 by dpyka