OpenRoberta/robertalab #29: make programs between robot and simulation exchange able

ensonic, 2016-01-22:
Teachers would love to let kids try the simulation and the send the code the wrote to the actual robot. This would also fix the sub-optiomal UI that one gets right now, when having an unsaved program in the simulator and switching to the robot.

See issue #28 for a pre-requisite.

bjost2s, 2016-02-25:
we will start working on this. The proposed solution is:

- Using all blocks from EV3 in simulation -> no extra toolbox needed anymore

- The user can start the program either on the robot or in the simulation -> No switching activity necessary anymore

- The simulation will always execute the program based on the standard configuration for the ev3 -> user will be warned if they have changed the configuration before executing the program in the simulation

- It will be possible to leave the window open from the simulation and change the program, if desired 

detailed tickets will be created

bjost2s, 2016-04-06:
done

closed 2016-04-06 by bjost2s