OpenRoberta/robertalab #30: provide a tool to manage sdcard images

ensonic, 2016-01-22:
In order to run openroberta lab with ev3s one needs to first prep sdcard images for them and then setup wifi. For class room setups this still takes some times for preparation (setting up the wifi password 10 times).
In addition for classrooms that will use a local roberta server, the teacher will need to setup and manage the raspberry pi 2 server (another sdcard image to fetch and also wifi to setup).

What if we write a portable app (java) that makes this easy?

* user selects which of the scenarious is in use:
  1. only ev3 images using lab.openroberta.org
  2. ev3 images using a local server
* user configured wifi settings (ssid and password)
* the app remembers what images one has downloaded last and can check if updates are available
* when updating, the app downloads the images and applies customization
  * scenario 1:
    * set the wifi configuration on the ev3 images
    * set unique hostnames for each ev3 image (just number them or use real names from a list)
    * select lab.openroberta.org as the server
  * scenario 2: (in addition to scenario 1)
    * set the openrobertalab.local as the server
    * configure wifi access point on the raspi image
* then a dialog will ask the user to insert cards, write them, insert next card, ... until all is up-to-date
* there could be some menu item to rewrite a ev3 image or raspi image (e.g. if the card was lost or got broken).

For the future the tool could also handle doing database setups of the raspi and restoring those after sd-card updates.

bjost2s, 2016-01-24:
I really like the idea and we should have that by release 2.0 where the user has to create a new image for jejos anyway, so I add milestone 2.0 to this ticket

ensonic, 2016-03-04:
We can actually use libraries like: https://github.com/gkostka/lwext4 to modify the image. Another options is to just patch the wifi password in the image.

ensonic, 2016-09-27:
This tool looks promissing: https://www.etcher.io/