OpenRoberta/robertalab #31: Could not resolve dependencies for project ...

smackoz, 2016-01-27:
Newbie needing some assistance.

Installed jdk-7u55-macosx-x64, JAVA_HOME set correctly.
Installed apache-maven-3.3.9 to /opt, exported /opt/apache-maven-3.3.9/bin to PATH
Installed phantomjs-2..1.1-linux-x86_64 to /opt, exported /opt/phantomjs-2.1.1-linux-x86_64/bin to PATH

Invoked mvn clean install in OpenRobertaParent and this is as far as I can get.
How to I resolve the error below?
Thanks in advance for you help.

[INFO] Reactor Summary:
[INFO] 
[INFO] RobertaParent ...................................... SUCCESS [ 45.351 s]
[INFO] Resources .......................................... SUCCESS [  0.009 s]
[INFO] OpenRobertaShared .................................. SUCCESS [05:51 min]
[INFO] OpenRobertaRuntime ................................. FAILURE [ 35.643 s]
[INFO] EV3Menu ............................................ SKIPPED
[INFO] OpenRobertaRobot ................................... SKIPPED
[INFO] OpenRobertaServer .................................. SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 07:13 min
[INFO] Finished at: 2016-01-27T21:13:31+11:00
[INFO] Final Memory: 22M/148M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal on project OpenRobertaRuntime: Could not resolve dependencies for project de.fhg.iais.openroberta:OpenRobertaRuntime:jar:1.3.2-SNAPSHOT: The following artifacts could not be resolved: de.fhg.iais.roberta:ev3classes:jar:0.9.0-beta, de.fhg.iais.roberta:dbusjava:jar:0.9.0-beta, org.lejos:jna:jar:3.2.7: Could not find artifact de.fhg.iais.roberta:ev3classes:jar:0.9.0-beta in ORA-releases (http://mp-devel.iais.fraunhofer.de:8081/nexus/content/repositories/ORA-releases/) -> [Help 1]

ensonic, 2016-01-27:
The http://mp-devel.iais.fraunhofer.de:8081/nexus/content/repositories/ORA-releases/ is an old url for a maven repo. Is this an up to date git clone? Which branch did you checkout (develop?) If you cloned master, please checkout the develop branch.

smackoz, 2016-01-27:
Thanks, successfully compiled!

[INFO] RobertaParent ...................................... SUCCESS [ 56.712 s]
[INFO] Resources .......................................... SUCCESS [  0.099 s]
[INFO] OpenRobertaShared .................................. SUCCESS [ 10.287 s]
[INFO] OpenRobertaRuntime ................................. SUCCESS [ 40.056 s]
[INFO] EV3Menu ............................................ SUCCESS [ 18.043 s]
[INFO] OpenRobertaRobot ................................... SUCCESS [05:35 min]
[INFO] OpenRobertaServer .................................. SUCCESS [03:14 min]
[INFO] OpenRobertaUSB ..................................... SUCCESS [ 13.680 s]
[INFO] ------------------------------------------------------------------------
[INFO] BUILD SUCCESS
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 11:09 min
[INFO] Finished at: 2016-01-27T23:15:05+11:00
[INFO] Final Memory: 46M/262M

closed 2016-01-27 by smackoz

dpyka, 2016-01-27:
It should be possible to build the master branch again. Master is now also using the ok-nexus server, instead of the old mp-devel.