OpenRoberta/robertalab #32: token for pairing should be generated on the server

ensonic, 2016-01-29:
Right now the token is generated on the ev3. This has two issues:
when scaling this to more devices, we need to replicate the code
we cannot to 100% avoid code collisions.
I propose to change the protocol:
  1. the robot sends the 'register' without at token
  2. reply contains unique code

We can keep support for the old method for a while. That is the server will use the client provided token if there is one and otherwise generate and provide one.

bjost2s, 2016-02-01:
In the future we would like to support robots without a display. These robots have to send a unique name, id or mac adress, which is somehow visible on the robot (eg. sticker). In this case, there is nothing to do on server side, I think. So they have to register with a "token".

ensonic, 2016-02-02:
Good point. Also after inspecting the sources, generating the code on the server side, makes the workflow more complicated on the client side, since we need to first connect and then when we get the token from the server signal the UI to show the token.

bjost2s, 2016-04-06:
close?

closed 2016-04-07 by ensonic