OpenRoberta/robertalab #38: NXT integration

dpyka, 2016-02-10:
We want to support the NXT using the standard fimware. If it is possible, we want to use the Java USB program for the connection.
Generating NXC code and compile it with the NBC compiler. Executable for Windows and Linux, working out of the box.

Linux
Install libusb-0.1 or libusb-compat if libusb-1.0 is already installed (NXT only supports this ancient version).
Windows
Older systems use the Fantom driver, but this driver is not signed and will not work for Windows 8 upwards. Check if we can use libusb for Windows as well.

Lejos already has a lot of PC tools which also work for the standard firmware. Reading brick information like name, battery, firmwareversion etc. is supported as well as uploading and executing files. Lejos is using libusb under Linux but the Fantom driver under Windows. If we can change that to always use libusb, we would get the whole communication part almost for free.

dpyka, 2016-07-22:
The lejos usb implementation refers to some c code via Java Native Interface. I added the missing files (test if filenames are correctly referenced in the code!) and also the udev rules which will be required for the installer under Linux.

I think there are several libusb and also fantom driver versions now, which may not work with the leJOS implementation. We have to check and find at least one pair of driver + usb program implementation that is actually working.

cvejoski, 2016-08-01:
Additionally if a Ubuntu is used we need to install libusb-dev and ant for compiling the driver. Then copy the .so file to src/main/resources/ in the USB project

closed 2016-09-21 by cvejoski