OpenRoberta/robertalab #42: one needs to select "new" twice to discard a program

ensonic, 2016-02-12:
When I select "new" from the menu I get the:

'''
Attention
You have unsaved changes in your program or configuration. Sign in and save your program or configuration.
'''
warning with an "ok" button. When I click "ok"  the code is still there. I then have to select "new" from the menu again and the code is discarded without warning.

I think the dialog should have two buttons "delete code" and "cancel", where "delete code" will proceed and "cancel" will get me back to the code.

bjost2s, 2016-02-25:
This should be fixed for switching from one robot system to another robot system. Switching from robot to simulation is no longer needed. We do not have a second robot in 1.4 but we should implement that as if we have one, to have the gui prepared for many robot systems.

ensonic, 2016-02-29:
This is not related to robot switching.

bjost2s, 2016-04-06:
done

closed 2016-04-06 by bjost2s