OpenRoberta/robertalab #43: when switching to the simulator the robot is not disconnected

ensonic, 2016-02-12:
Repro:
* start as ev3
* pair the ev3
* switch to sim
* note: the robot is still connected (on the robot side)
* switch  back to ev3
* the robot is still connected, but the ui shows it as disconnected
* when I reconnect the robot with the same token I get an exception:
https://gist.github.com/ensonic/fecbd222f41fc7530bbb

I have a fix:

ensonic, 2016-02-15:
Once the simulator matches the real robot, the question is if it wouldn't be better to retain the connection instead. This way you can switch back and forth without having to reconnect.

bjost2s, 2016-02-25:
Should be fixed with #29. I think right now the connection is still there, but the gui is ignoring the robots server state. If we remove the switching from robot to simulation anyway this problem shouldn't appear anymore.

ensonic, 2016-02-25:
We need to remember to revert my change then.

bjost2s, 2016-02-25:
We will need it, if one switch from one robot system to another, because then disconnecting is reqested.

bjost2s, 2016-04-06:
from my understanding this is done, please reopen, if not

commit a7650ffa (2016-02-17): Disconnect the previous robot when switching between ev3 and sim.

Otherwise the robot stays connected and if it tries to reconnect we get an
exception thrown.

Fixes #43
closed 2016-04-06 by bjost2s