OpenRoberta/robertalab #44: detect/check robot configuration

ensonic, 2016-02-15:
Right now we leave robot configuration to the user. If the session is connected to an ev3, we could actually send connection data along with each push request. Now the question is how can me make use of that in the UI.

1. we could add a "check" button that is the counterpart of the "check" button in the program tab. This would flag out-dated configuration entries.

2. we could have an auto-update checkbox. Then the configuration will auto update to reflect the latest config we got via push.

Once we figured that, I'd love to show the current sensor values / motor positions in the configuration too.

dpyka, 2016-02-15:
I prefer option 2 with auto-update by default. But we also have to make the brick configuration more visible to the user.

bjost2s, 2016-03-01:
option two is also for me a good solution

VinArt, 2019-02-01:
Does not work for all robots, e.g. Arduino.

closed 2019-02-01 by VinArt