OpenRoberta/robertalab #45: lab programming: functions don't reach global variables

alesolano, 2016-02-16:
Hello everyone!

Well, I have started using the Programming Environment http://lab.open-roberta.org/ and I can't get to use global variables in functions. The program does not compile. 

(I attach a screenshot of the code, if it makes the explanation simpler...)
[image]

I can trick a little the compiler if I create local variables inside the functions, passing the values of the global variables to the local variables. But, if I change the values of the local variables, there's no way to get these values out of the function.

Thank you!

P.S: I don't know if this is the correct place to post this issue.
P.S.2: Congratulations and thank you for this amazing project!

ensonic, 2016-02-16:
Yes, this is definitely the right place for the report. One quick question - are you using lejos or ev3dev on the ev3? Also if you can click the [< >] button on the bottom left you can check the generated code and posting that would help here.

alesolano, 2016-02-16:
I am using ev3dev.

And this is the code:
`package generated.main;

import de.fhg.iais.roberta.runtime.*;
import de.fhg.iais.roberta.runtime.ev3.*;

import de.fhg.iais.roberta.shared.*;
import de.fhg.iais.roberta.shared.action.ev3.*;
import de.fhg.iais.roberta.shared.sensor.ev3.*;

import de.fhg.iais.roberta.components.*;
import de.fhg.iais.roberta.components.ev3.*;

import java.util.LinkedHashSet;
import java.util.Set;
import java.util.List;
import java.util.ArrayList;
import java.util.Arrays;

import lejos.remote.nxt.NXTConnection;

public class goMaxDistance {
    private static final boolean TRUE = true;
    private static Ev3Configuration brickConfiguration;

    private Set<EV3Sensors> usedSensors = new LinkedHashSet<EV3Sensors>(Arrays.asList(EV3Sensors.EV3_ULTRASONIC_SENSOR));

    private Hal hal = new Hal(brickConfiguration, usedSensors);

    public static void main(String[] args) {
        try {
             brickConfiguration = new Ev3Configuration.Builder()
                .setWheelDiameter(5.6)
                .setTrackWidth(18.0)
                .addActor(ActorPort.B, new EV3Actor(EV3Actors.EV3_LARGE_MOTOR, true, DriveDirection.FOREWARD, MotorSide.RIGHT))
                .addActor(ActorPort.C, new EV3Actor(EV3Actors.EV3_LARGE_MOTOR, true, DriveDirection.FOREWARD, MotorSide.LEFT))
                .addSensor(SensorPort.S1, new EV3Sensor(EV3Sensors.EV3_TOUCH_SENSOR))
                .addSensor(SensorPort.S4, new EV3Sensor(EV3Sensors.EV3_ULTRASONIC_SENSOR))
                .build();
            new goMaxDistance().run();
        } catch ( Exception e ) {
            lejos.hardware.lcd.TextLCD lcd = lejos.hardware.ev3.LocalEV3.get().getTextLCD();
            lcd.clear();
            lcd.drawString("Error in the EV3", 0, 0);
            if (e.getMessage() != null) {
                lcd.drawString("Error message:", 0, 2);
                Hal.formatInfoMessage(e.getMessage(), lcd);
            }
            lcd.drawString("Press any key", 0, 7);
            lejos.hardware.Button.waitForAnyPress();
        }
    }

    float maxDistance = 0;
    float iterationMaxDistance = 0;
    float iteration = 0;

    public void run() {

        findMaxDistance();
        pointToMaxDistance();
        hal.regulatedDrive(ActorPort.C, ActorPort.B, false, DriveDirection.FOREWARD, 30);
        hal.waitFor(4000);
        hal.stopRegulatedDrive(ActorPort.C, ActorPort.B);
    }
        
    private void findMaxDistance() {
        if ( TRUE ) {
            while ( iteration <= 15 ) {
                hal.rotateDirectionRegulated(ActorPort.C, ActorPort.B, false, TurnDirection.RIGHT, 10);
                hal.waitFor(500);
                if ( hal.getUltraSonicSensorDistance(SensorPort.S4) >= maxDistance ) {
                    maxDistance = hal.getUltraSonicSensorDistance(SensorPort.S4);
                    iterationMaxDistance = iteration;
                }
                iteration = iteration + 1;
            }
        }
    }
        
        
    private void pointToMaxDistance() {
        for ( float k0 = 0; k0< 15 - iterationMaxDistance; k0+= 1 ) {
            hal.rotateDirectionRegulated(ActorPort.C, ActorPort.B, false, TurnDirection.LEFT, 10);
            hal.waitFor(500);
        }
    }
        
        
        
        
        
        
}
`

ensonic, 2016-02-16:
This is java code. That means either you are using lejos or you did not connect to your robot (in which case it does not know that you might use ev3dev).

alesolano, 2016-02-16:
Ok, I have the code in a proper way now:

```
#!/usr/bin/python

from __future__ import absolute_import
from roberta.ev3 import Hal
from roberta import BlocklyMethods
from sets import Set
from ev3dev import ev3 as ev3dev
import math

TRUE = True
brickConfiguration = {
    'wheel-diameter': 5.6,
    'track-width': 18.0,
    'actors': {
        'A':Hal.makeLargeMotor(ev3dev.OUTPUT_A, 'on', 'foreward', 'right'),
        'B':Hal.makeLargeMotor(ev3dev.OUTPUT_B, 'on', 'foreward', 'left'),
    },
    'sensors': {
        '4':ev3dev.UltrasonicSensor(ev3dev.INPUT_4),
    },
}
usedSensors = Set(['EV3Sensors.EV3_ULTRASONIC_SENSOR'])
hal = Hal(brickConfiguration, usedSensors)

def run():
    maxDistance = 0
    iterationMaxDistance = 0
    iteration = 0

```

But I have had to copy the code from the brick itself, using puTTY. Open Roberta Lab only gives me java code.

And this is the error that the brick shows me. (The bold line also appears on the brick display screen)
> robot@ev3dev:~$ ./goMaxDistance.py
> 
> Traceback (most recent call last):
>   File "./goMaxDistance.py", line 73, in <module>
>     main()
>   File "./goMaxDistance.py", line 63, in main
>     run()
>   File "./goMaxDistance.py", line 29, in run
>     findMaxDistance()
>   File "./goMaxDistance.py", line 38, in findMaxDistance
>     while iteration <= 15 :
> UnboundLocalError: **local variable 'iteration' referenced before assignment**

ensonic, 2016-02-17:
Thanks! This is indeed wrong. I'll fix this right away so that we have an update for the upcomming server release.

ensonic, 2016-02-17:
Fixed in commit 1ea1a4e240d3ee88befd841f5f560850989f7296

closed 2016-02-17 by ensonic

alesolano, 2016-02-17:
:beers::beers: Cool!
So glad to help a little in the project!