OpenRoberta/robertalab #46: add documentation for how to build blockly

ensonic, 2016-02-17:
It is not obvious that changes made on blockly are not taken into acount when building. So what about mentioning this in the top-level README and also tell people what they have to do to build it.

What I tried so far:
download the closure-library as stated on http://code.google.com/p/blockly/wiki/Closure
revert changes in blockly/build.py
https://gist.github.com/ensonic/e6ca1cd7cd0d0845bc93

Then it builds, but when I start the server I get:
http://picpaste.de/pics/Screenshot_from_2016-02-17_13_55_16-HhGO6qrg.1455713744.png

ensonic, 2016-02-22:
I've retried this after the blockly update:
```
cd OpenRobertaServer/staticResources/blockly && python build.py
...
SUCCESS: blocks_compressed.js
Size changed from 300 KB to 149 KB (50%).
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "build.py", line 177, in run
    self.gen_generator("javascript")
  File "build.py", line 252, in gen_generator
    f = open(filename)
IOError: [Errno 2] No such file or directory: 'generators/javascript.js'
```

bjost2s, 2016-02-23:
Please have a look at blockly. As we do all changes in our robertalab/blockly master branch for now, python build.py will work as descriped in google/blockly. There is now one new feature, the roberta messages: Please add new blockly messages in robMsg/robMessages.js. This will automatically merged in the blockly messages.js. It helps to separate the messages for the translation process.

ensonic, 2016-02-24:
We don't have Issues enabled for https://github.com/OpenRoberta/blockly. Hence I've filed it here.
Still with the new setup, I have to do this patch, since we don#t have the generators in git:
``` patch
> git diff build.py
diff --git a/OpenRobertaServer/staticResources/blockly/build.py b/OpenRobertaServer/staticResources/blockly/build.py
index 87841cb..9fcc0f6 100755
--- a/OpenRobertaServer/staticResources/blockly/build.py
+++ b/OpenRobertaServer/staticResources/blockly/build.py
@@ -174,10 +174,10 @@ class Gen_compressed(threading.Thread):
   def run(self):
     self.gen_core()
     self.gen_blocks()
-    self.gen_generator("javascript")
-    self.gen_generator("python")
-    self.gen_generator("php")
-    self.gen_generator("dart")
+    #self.gen_generator("javascript")
+    #self.gen_generator("python")
+    #self.gen_generator("php")
+    #self.gen_generator("dart")
 
   def gen_core(self):
     target_filename = "blockly_compressed.js"
```

bjost2s, 2016-02-25:
ok, I understand. Issues will be enabled for OpenRoberta/blockly when we closed #18 (maybe today). And the workflow should be that changes in blockly are commited to OpenRoberta/blockly, after that these changes have to be copied by hand into the robertal and have to be commited their again.

ensonic, 2016-02-25:
It should be enough to push changes to OpenRoberta/blockly and then updating the submodule on the OpenRoberta/robertalab

ensonic, 2016-04-22:
This is still happening - are you just ignoring the error?
```
> cd OpenRobertaServer/staticResources/blockly && python build.py
python: can't open file 'i18n/js_to_json.py': [Errno 2] No such file or directory
('Error running i18n/js_to_json.py: ', CalledProcessError())
SUCCESS: blockly_uncompressed.js
SUCCESS: blockly_compressed.js
Size changed from 2566 KB to 614 KB (24%).
SUCCESS: blocks_compressed.js
Size changed from 301 KB to 150 KB (50%).
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "build.py", line 177, in run
    self.gen_generator("javascript")
  File "build.py", line 252, in gen_generator
    f = open(filename)
IOError: [Errno 2] No such file or directory: 'generators/javascript.js'
```

cvejoski, 2016-04-22:
The build of the blockly is only done in the OpenRoberta/Blockly project and then copied to the OpenRobertaServer/StaticResources. You can not build Blockly in OpenRobertaServer project

closed 2016-04-25 by bjost2s