OpenRoberta/robertalab #50: Inform the user, if the execution of a program on the robot fails

rbudde, 2016-02-23:
If the program uploaded to the robot fails "early" (e.g. compiled with an incompatible javac), the failure is simply ignored. A notification should be sent to the server, sent to the client and displayed there.

dpyka, 2016-02-25:
There is a new field nepoexitvalue in the ev3communicationdata to tell the frontend if there are problems executing the NEPO program.

Exitcodes are for lejos:
* 0 everything is fine
* 143 process killed by the menu (button down and enter pressed)

@cvejoski please provide a popup message with the following behaviour:
Check if the exitvalue has changed from the last ping
Check if it is not 0 and not 143, display the error message

dpyka, 2016-02-25:
@ensonic please have a look at my commit for implementing this. The error codes may be different for the ev3dev

ensonic, 2016-02-25:
@dpyka if this commit extends the protool, can you please update
https://github.com/OpenRoberta/robertalab/wiki/Robot-Server-Protocol

ensonic, 2016-04-13:
I think this is done.

closed 2016-04-13 by ensonic