OpenRoberta/robertalab #51: Connecting bug

dpyka, 2016-02-25:
[image]
[image]
[image]
[image]

Original email:

Sehr geehrte Mitarbeiter von Open-Roberta,
 
seit ca. drei Wochen habe ich häufiger Probleme damit, EV3 Bricks via W-LAN mit open-roberta zu verbinden. 
 
Bei 5 von 7 Bricks lies sich das Programm nach Anweisung installieren und auch das einwählen ins W-LAN verlief problemlos. Als jedoch die Bricks mit der open-roberta Plattform verbunden werden sollten, kam ein Ton nachdem ca. 2-3 Sekunden der Verbindungscode angezeigt wurde und es erschien eine Art Hinweis, dass ein Programmierungsfehler vorliege. Es dürfe kein "{" benutzt werden oder so ähnlich. 
Daraufhin fährt sich der Brick wieder runter. Bei Wiederholung geschieht das gleiche. 
Wo könnte da das Problem liegen? 
 
Ich setze dann meistens die SD-Karte neu auf und in ca. 60% der Fälle funktioniert es dann. Allerdings tritt das Problem auch auf, wenn sich der Roboter im Vorfeld erfolgreich mit open-roberta verbunden hatte, wir aber dann eine Woche später weiterarbeiten möchten.
Das ist sehr unangenehm, weil die Kinder dann auch nicht weiterarbeiten können. 
 
Weiterhin habe ich mit zwei Bricks das Problem, dass LeJos nicht installiert wird. Nach kurzer Installationszeit sieht man oben links neben dem L ein "-" aufblinken und es passiert nichts mehr. Ich habe schon sehr lange gewartet, aber es ändert sich nicht. 
Auch das neue Aufsetzen der SD-Karte bringt keine Veränerung. 
 
Können Sie mir bei diesen Problemen weiterhelfen? 
 
Vielen Dank im Voraus! 
Mit freundlichen Grüßen 
M. Acht

dpyka, 2016-02-25:
Seems like the EV3 is sending invalid json data and the server response is empty or error. Maybe because of some special characters in the brick name?

dpyka, 2016-02-26:
Also unsere EV3s heißen Alle Rotti#, wobei # für die Nummer der Gruppe steht. Soll ich es mal ohne Sonderzeichen ausprobieren? 
Als WLAN-Dongle nutze ich den Racksoy Professionell Ralink 5370 Wifi Dongle USB 2.0 Wireless 802.11n USB Wifi Adapter Kompatibel mit Raspberry Pi, Windows 2000/XP/VISTA/Win7/8/Linux 2.4 oder Höher.
Bei allen anderen außer diesen beiden Bricks tunktionieren sie einwandfrei. 
Vielleicht nutzen diese Infos ja etwas.

dpyka, 2016-02-26:
Still i am not able to reproduce the bug from this guy.

I just came across another bug. If the mac address is not readable on the ev3 for whatever reason, it will send "unknown" to the server. If there are at least two bricks with "unknown" mac address, they will delete each others sessions when trying to register with the server.

bjost2s, 2016-04-25:
still a bug?

closed 2016-10-13 by bjost2s