OpenRoberta/robertalab #56: Apply suggested fixes reported by pagespeed

ensonic, 2016-02-26:
https://developers.google.com/speed/pagespeed/insights/
I still have this installed as a chrome extension which now is deprecated :/ There is a variant of this in the chromestore too which I haven't tested.
https://chrome.google.com/webstore/detail/pagespeed-insights-with-p/lanlbpjbalfkflkhegagflkgcfklnbnh

The downside of the online version is that one cannot analyze the current local dev version. We can also potentially integrate this as a test: https://github.com/addyosmani/psi

Anyway the tool points out some interesting ways to optimize the page loading speed for the lab. Right now we have a score of 13 test cases passing and 12 open suggestions:

- [ ] Minimize payload
  - [ ] **H** Enable compression
  - [ ] **H** Serve scaled images
  - [ ] **H** Minify JavaScript
  - [ ] **M** Serve resources from a consistent URL
  - [ ] **M** Optimize images
  - [ ] **L** Minify HTML
  - [ ] **L** Minify CSS
- [ ] Minimize delay in page load 
  - [ ] **L** Specify image dimensions
  - [ ] **L** Specify a character set
- [ ] Other 
  - [ ] **H** Leverage browser caching
  - [ ] **H** Specify a cache validator
  - [ ] **L** Defer parsing of JavaScript
  - [ ] **L** Specify a Vary: Accept-Encoding header

ensonic, 2016-06-08:
@rbudde I have this patch to enable gzip compression
https://gist.github.com/ensonic/f47946a07941545cda12336ff9509f35
This will compress files as we sent them and thus create load on the server.
There should be also a way to enable gzip compression for jetty on the DefaultServlet by setting the "gzip" init-parameter, but I could not find the related API. That would also require to have a maven rule to gzip static content.
See: https://wiki.eclipse.org/Jetty/Feature/GZIP_Compression

drigz, 2016-07-15:
#176 should improve this by compressing JS and CSS. I haven't enabled minification for either - it should be harmless for CSS but might break the JS.

Here is the current breakdown of page size with the compression:

- Total: 807 KB
- Javascript: 454 KB (could be improved with an optimizer, eg [r.js](http://requirejs.org/docs/optimization.html))
- CSS: 33 KB
- Images: 155 KB (mostly already compressed, <5 KB SVGs)
- Sounds: 15KB
- Fonts: 83 KB (typicons.ttf, could be gzipped)
- HTML: 44KB (all index.html, could be gzipped)
- Mystery: 23 KB (overhead? not in Chrome network tab)

ensonic, 2016-07-18:
"Enable compression" is now a "M" severity due to the index,html and the css/svg/NEPO_Logo_RGB.svg

drigz, 2016-07-18:
For "Serve resources from a consistent URL": https://api.jquery.com/jquery.getscript/#caching-requests

VinArt, 2018-11-23:
@ensonic currently it is possible to analyse latest develop, because the test server is running it

VinArt, 2019-02-01:
Compression is done, complete solution will be available when using webpack

closed 2019-02-01 by VinArt