OpenRoberta/robertalab #60: Mathematik - Division nicht eindeutig

bjost2s, 2016-02-29:
Wenn zwei ganzzahlige Parameter eingegeben werden, wird eine Ganzzahldivision durchgeführt, also z.B. 50/90 = 0. Ist einer der Parameter eine Variable, dann erfolgt eine „normale“ Dezimaldivision. Die angehängte Bilddatei zeigt das Problem.
[image]

bjost2s, 2016-02-29:
Is it also an issue on ev3dev?

ensonic, 2016-02-29:
On ev3dev it is both times 0
```python
bla = 90
hal.drawText(str(50 / 90), 0, 0)
hal.drawText(str(50 / bla), 0, 2)
 ```
Even worse is that we don't have a way to 'enforce' float math. I can't enter '''90.0''' for the value to show that it is a float; it will be reduced to '''90'''. Maybe we need to have two number blocks; one for int and one for float and alias the old number to int?

bjost2s, 2016-02-29:
We check on client side the  decimal mark, why not simply add a ".0" there. when sending to the server. all computed results will have a decimal mark then by default.

ensonic, 2016-02-29:
That's what I said. I've tried to use ```90.0```, but something strips the ```.0```. We should not do that for a start.

bjost2s, 2016-04-25:
is it fixed? I'm not sure

cvejoski, 2016-06-02:
@ensonic can you please check if everything is fine on the ev3dev side with the division

ensonic, 2016-06-03:
First one gives 0.555555556 on ev3dev.

closed 2017-03-21 by cvejoski