OpenRoberta/robertalab #61: Der Experten-Block Text - Erstelle Text aus arbeitet nicht korrekt.

bjost2s, 2016-02-29:
Wenn am Eingabeslot ein mathematischer/logischer Ausdruck steht, wird dieser nur korrekt verarbeitet, wenn er am ersten Slot angedockt wird. Falls ein anderer als der erste Slot einen Ausdruck erhält, gibt es eine Fehlermeldung (siehe angehängte Datei).
[image]

bjost2s, 2016-02-29:
Please check if this is also an issue on ev3dev

ensonic, 2016-02-29:
Jeps, also broken on ev3dev with python:
```
BlocklyMethods.textJoin("17 + 4 = "; 17 + 4)
```

ensonic, 2016-02-29:
Hmm, there is probably something wrong with the AST:
https://github.com/OpenRoberta/robertalab/blob/19a58ba2fd5185bad4ab4e470c380bcaca682a5b/OpenRobertaRobot/src/main/java/de/fhg/iais/roberta/syntax/codegen/ev3/Ast2Ev3JavaVisitor.java#L1249-L1256

The loop is always just iterated once with expr being an ArrayList with the params. That also explains why the visitExprList is called.

cvejoski, 2016-03-01:
There was bug (strange code :)) in the implementation of the AST transformer for the text join block.

closed 2016-03-01 by cvejoski