OpenRoberta/robertalab #63: NEPO rechnet komplexere Rechnungen möglicherweise nicht korrekt.

bjost2s, 2016-02-29:
Beispiel: 60 / 180 * pi
geklammert: (60 / 180) * pi = 1,0472
falsch geklammert: 60 / (180 * pi) = 0,1061
NEPO gibt für beide geklammerten Rechnungen das gleiche Ergebnis aus, nämlich 1,0472.
Programm im Anhang
[image]

bjost2s, 2016-02-29:
Please check if this is also an issue on ev3dev

cvejoski, 2016-03-01:
This is a problem in general because we do not know when to use parentheses. We can add parentheses with every math expression like Blockly is doing.

bjost2s, 2016-03-01:
As a user I would expect that each block is surrounded by parentheses, similar to blockly

bjost2s, 2016-04-06:
Kostadin, can you fix this for 1.4?

cvejoski, 2016-05-17:
This is fixed for 1.5. We wrap nested binary expressions with parentheses.

closed 2016-05-17 by cvejoski

ensonic, 2016-05-18:
@cvejoski was it does for all code generators?

cvejoski, 2016-05-18:
@ensonic this is done for all but I have only pushed for java and javascript, for python I have changed it but I have not pushed the commit

ensonic, 2016-05-18:
@cvejoski I can't test it right now, but maybe it is obvious. Could you paste the diff to a gist?

ensonic, 2016-06-01:
Looks good to me, please push.

cvejoski, 2016-06-01:
I have already pushed the changes for the python code generator

ensonic, 2016-06-03:
Works, thanks!