OpenRoberta/robertalab #64: Math - modes of list

bjost2s, 2016-02-29:
This block with the "modes" setting does nothing but returning the list that was given as input value.
[image]

bjost2s, 2016-02-29:
The "mode" is copied from blockly, but there, it is not really clear what the result should be (generating always [0,1,3])

-> +1 for deleting this drop down value

ensonic, 2016-03-17:
For reference: this is the js impl
https://github.com/google/blockly/blob/master/generators/javascript/math.js#L288

The "mode" is the value that occurs most often. If no number is repeated, then there is no mode for the list.

bjost2s, 2016-03-18:
as discussed: will be removed see OpenRoberta/blockly#9

bjost2s, 2016-04-06:
fixed

closed 2016-04-06 by bjost2s