OpenRoberta/robertalab #65: New block - driving in a curve

bjost2s, 2016-02-29:
I think there should be a block which combines rotating and driving (like the block "Lenkung") at Legos programming tol

cvejoski, 2016-09-20:
EV3 implementation: bab13c48ae3032661fd82bf8a18a3828be499a1d

closed 2016-09-21 by bjost2s

ensonic, 2016-09-23:
A few questions:
1) why don't we pass the motor ports to the hal like we do for all the other drive commands.
2) In https://github.com/OpenRoberta/robertalab/commit/bab13c48ae3032661fd82bf8a18a3828be499a1d the optional parameter is MotorDuration<Void> duration = curveAction.getParamLeft().getDuration(); but on the hal side https://github.com/OpenRoberta/robertalab-ev3lejos/commit/320ec18727faf007e13d5268b3ce03cc72e1fc90 it is called "float distance"? A duration would be weird since the duration also depends on the gradient of the underground. A distance is weird too since the outer wheel would drive further than the inner wheel. Should this maybe be a angle?

cvejoski, 2016-09-23:
1) We do not pass the motor ports for the drive commands, since this differential drive and we need only left and right motor and that we can get from the configuration.

2) For curveAction.getParamLeft().getDuration(); you are right. This duration is used for the drive blocks to store the information **distance** or **rotations** the robot should make. Since we use only drive in curve for some distance the name of the parameter is distance. I will change it to **duration** so we stay consistent.

ensonic, 2016-09-23:
@cvejoski the block says distance in cm, I think it should be distance and not duration. Imho duration make not a lot of sense for motors with encoders. We measure the turns the take to get a precise distance. If you want duration, you don't pass a distance and then add a Wait(500 ms), right after.

elhele, 2016-09-23:
@ensonic it is also different for the different robots. In Bot'n Roll it is exactly duration in seconds, because this robot's motors don't have the encoders.

ensonic, 2016-09-23:
@elhele good point. Since the block UI shows distance=20cm what are we doing for e.g. the "Bot'n Roll"? Do we convert the cm into duration based on average speed?

elhele, 2016-09-23:
@ensonic in Bot'n Roll the block UI shows seconds, not: 
[image]
Calculation based on average speed would not be accurate, since it strongly depends on the battery charge, so we decided to change the block so it would just use the time and not the distance.

The robot is generally meant more for  tasks those require reactions to some sensor data, e.g. line following or obstacle avoidance, not for doing something for a certain amount of time. The library and sensor configuration were designed accordingly and it is hard to adjust some of our blocks as they were.