OpenRoberta/robertalab #68: Simple program not running on ev3dev - configuration problem

bjost2s, 2016-02-29:
When starting a simple program on ev3dev with no sensors attached the robot does nothing
- no error message
- no action
- blank display for a short time

it is requested to let the robot exe
cute everything which is not against the configuration, like with lejos
[image]

ensonic, 2016-02-29:
Did you run:
```
sudo apt-get update
sudo apt-get dist-upgrade
```
on the ev3 recently? I released a new package last week that handles the exceptions regarding not conencted sensors in the config.

bjost2s, 2016-03-01:
Thanks, this works, but it took more than one hour to execute this two commands!

closed 2016-03-01 by bjost2s

dlech, 2016-03-02:
The update took so long because debian did a point release since the last ev3dev image was release, so many packages were updated. In the future, to only update one package, you can run

    sudo apt-get update
    sudo apt-get install openrobertalab