OpenRoberta/robertalab #69: Program with unregulated motors causes error:

bjost2s, 2016-02-29:
Fehler im EV3  press any key

[image]

[image]

ensonic, 2016-03-01:
This was before you updated the libraries, right. Please reopen if still happens. I just tries it and it works.

bjost2s, 2016-03-02:
Not working for me :-(. Maybe I have to update it again?

ensonic, 2016-03-02:
I have done a lot of fixes and will push another update next week. I've tested it and it works. There was another bug in the codegen, that missed a quote for the "motor port x on speed" block. I need more tests for the codegen :)

closed 2016-03-02 by ensonic