OpenRoberta/robertalab #70: Disconnecting - Reconnection not possible

bjost2s, 2016-02-29:
when disconnecting the robot from the lab, the robot does not erase the roberta symbol in the menu. This is somehow misleading for the user.
After connecting again (with a new token) i get the following log and the robot is not connected:
21:05:55.110 [qtp1508256644-34] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 0 msec: head navigation menu item clicked
21:06:22.763 [qtp1508256644-16] INFO d.f.i.r.r.ev3.Ev3CommunicationData - BRICK ev3dev [22:16:53:40:c4:f8] starts waiting for the client to approve a token
21:06:22.765 [qtp1508256644-34] INFO d.fhg.iais.roberta.util.ClientLogger - 1 log entry written
21:06:22.768 [qtp1508256644-38] INFO d.f.i.r.r.ev3.Ev3CommunicationData - user approved the token. The approval request was scheduled 0 secs ago
21:06:22.768 [qtp1508256644-38] INFO d.f.i.r.r.ev3.Ev3Communicator - token X4R9Z872 is approved by a user.
21:06:22.768 [qtp1508256644-38] INFO d.f.i.r.j.r.all.ClientAdmin - success: token X4R9Z872 is registered in the session
21:06:22.768 [qtp1508256644-12] INFO d.f.i.r.r.ev3.Ev3CommunicationData - user approval lost. Nobody is waiting. The approval request was scheduled 0 secs ago
21:06:22.773 [qtp1508256644-12] INFO d.f.i.r.r.ev3.Ev3Communicator - token X4R9Z872 is approved by a user.
21:06:22.773 [qtp1508256644-12] INFO d.f.i.r.j.r.all.ClientAdmin - success: token X4R9Z872 is registered in the session
21:06:22.773 [qtp1508256644-16] INFO d.f.i.r.r.ev3.Ev3CommunicationData - Robot request for token approval terminated SUCCESSFULLY. Time elapsed: 5
21:06:25.111 [qtp1508256644-13] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 5 msec: set token 'X4R9Z872'
21:06:25.113 [qtp1508256644-13] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 3 msec: set token 'X4R9Z872'
21:06:25.113 [qtp1508256644-13] INFO d.fhg.iais.roberta.util.ClientLogger - 2 log entries written
21:06:28.111 [qtp1508256644-34] INFO d.f.i.r.j.r.all.ClientPing - /ping [count:400]
21:06:28.111 [qtp1508256644-34] INFO d.f.i.r.r.ev3.Ev3Communicator - token X4R9Z872 disconnected

ensonic, 2016-03-01:
We also use two states of the symbol. There is the connected and the disconnected state When I disconnect, the icon changed immediately. The actual disconnect might take a while since it has to wait for the push request to timeout (this is a blocking operation :/).

You said "when disconnecting the robot from the lab", did you selected "Disconnect" on the ev3 or did you do it differently?

bjost2s, 2016-03-01:
The robot was connected via Wifi and I selected "disconnect" in the menu of the ev3 . As the GUI does not get any information in addition to what the server sends via the ping, the immidiate switch in the GUI can only be by change (disconnect and ping close to each other).

bjost2s, 2016-03-03:
I tried it again and here is the log from the server. The robot does not create a new token and the server is reacting in the following way:
06:47:09.721 [qtp1433292785-15] INFO d.f.i.r.j.r.ev3.Ev3Command - pushcmd - brick sends token QDR25DDG for registration
06:47:09.724 [qtp1433292785-15] INFO d.f.i.r.r.ev3.Ev3CommunicationData - BRICK ev3dev [22:16:53:40:c4:f8] starts waiting for the client to approve a token
06:47:24.861 [qtp1433292785-33] INFO d.f.i.r.j.r.all.ClientUser - command is: clear
06:47:24.944 [qtp1433292785-14] INFO d.f.i.r.j.r.all.ClientToolbox - command is: loadT
06:47:24.970 [qtp1433292785-14] INFO d.f.i.r.p.AbstractProcessor - Success: TOOLBOX_GET_ONE_SUCCESS
06:47:25.050 [qtp1433292785-34] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 112 msec: page init
06:47:25.050 [qtp1433292785-34] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] result from server: {"cmd":"clear","rc":"ok","serverTime":"Thu Mar 03 06:47:24 CET 2016"}
06:47:25.050 [qtp1433292785-34] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 1 msec: clear user
06:47:25.050 [qtp1433292785-34] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] result from server: {"cmd":"loadT","data":"<toolbox_set id=\"toolbox\" style=\"display: none\"><category name=\"TOOLBOX_ACTION\" svg=\"true\"> ...","rc":"ok","message":"ORA_TOOLBOX_GET_ONE_SUCCESS","cause":"ORA_TOOLBOX_GET_ONE_SUCCESS","serverTime":"Thu Mar 03 06:47:24 CET 2016"}
06:47:25.050 [qtp1433292785-34] INFO d.fhg.iais.roberta.util.ClientLogger - 4 log entries written
06:47:25.050 [qtp1433292785-34] INFO d.f.i.r.j.r.all.ClientAdmin - command is: setRobot
06:47:25.055 [qtp1433292785-16] INFO d.f.i.r.j.r.all.ClientToolbox - command is: loadT
06:47:25.058 [qtp1433292785-34] INFO d.f.i.r.j.r.all.ClientAdmin - set Robot: robot ev3 with id 42 already set
06:47:25.061 [qtp1433292785-16] INFO d.f.i.r.p.AbstractProcessor - Success: TOOLBOX_GET_ONE_SUCCESS
06:47:25.105 [qtp1433292785-33] INFO d.f.i.r.j.r.all.ClientToolbox - command is: loadT
06:47:25.109 [qtp1433292785-33] INFO d.f.i.r.p.AbstractProcessor - Success: TOOLBOX_GET_ONE_SUCCESS
06:47:25.120 [qtp1433292785-34] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] result from server: {"cmd":"setRobot","rc":"ok","message":"ORA_TOKEN_SET_SUCCESS","cause":"ORA_TOKEN_SET_SUCCESS","serverTime":"Thu Mar 03 06:47:25 CET 2016"}
06:47:25.121 [qtp1433292785-34] INFO d.fhg.iais.roberta.util.ClientLogger - 1 log entry written
06:47:25.121 [qtp1433292785-34] INFO d.f.i.r.j.r.all.ClientToolbox - command is: loadT
06:47:25.123 [qtp1433292785-34] INFO d.f.i.r.p.AbstractProcessor - Success: TOOLBOX_GET_ONE_SUCCESS
06:47:25.181 [qtp1433292785-16] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 1 msec: set robot 'ev3'
06:47:25.181 [qtp1433292785-16] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] result from server: {"cmd":"loadT","data":"<toolbox_set id=\"toolbox\" style=\"display: none\"><block type=\"robBrick_ultrasonic\"></block><bl ...","rc":"ok","message":"ORA_TOOLBOX_GET_ONE_SUCCESS","cause":"ORA_TOOLBOX_GET_ONE_SUCCESS","serverTime":"Thu Mar 03 06:47:25 CET 2016"}
06:47:25.181 [qtp1433292785-16] INFO d.fhg.iais.roberta.util.ClientLogger - 2 log entries written
06:47:25.181 [qtp1433292785-16] INFO d.f.i.r.j.r.all.ClientConfiguration - command is: loadC
06:47:25.188 [qtp1433292785-16] INFO d.f.i.r.p.AbstractProcessor - Success: CONFIGURATION_GET_ONE_SUCCESS
06:47:27.958 [qtp1433292785-34] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] result from server: {"cmd":"loadT","data":"<toolbox_set id=\"toolbox\" style=\"display: none\"><category name=\"TOOLBOX_ACTION\" svg=\"true\"> ...","rc":"ok","message":"ORA_TOOLBOX_GET_ONE_SUCCESS","cause":"ORA_TOOLBOX_GET_ONE_SUCCESS","serverTime":"Thu Mar 03 06:47:25 CET 2016"}
06:47:27.959 [qtp1433292785-34] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[INFO]] result from server: {"cmd":"loadT","data":"<toolbox_set id=\"toolbox\" style=\"display: none\"><category name=\"TOOLBOX_ACTION\" svg=\"true\"> ...","rc":"ok","message":"ORA_TOOLBOX_GET_ONE_SUCCESS","cause":"ORA_TOOLBOX_GET_ONE_SUCCESS","serverTime":"Thu Mar 03 06:47:25 CET 2016"}
06:47:27.959 [qtp1433292785-34] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 1 msec: start with ev3 clicked
06:47:27.959 [qtp1433292785-34] INFO d.fhg.iais.roberta.util.ClientLogger - 3 log entries written
06:47:33.958 [qtp1433292785-14] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 25 msec: head navigation menu item clicked
06:47:33.958 [qtp1433292785-14] INFO d.fhg.iais.roberta.util.ClientLogger - 1 log entry written
06:47:34.329 [qtp1433292785-35] INFO d.f.i.r.j.r.all.ClientAdmin - command is: setToken
06:47:34.329 [qtp1433292785-35] INFO d.f.i.r.r.ev3.Ev3CommunicationData - user approved the token. The approval request was scheduled 24 secs ago
06:47:34.329 [qtp1433292785-35] INFO d.f.i.r.r.ev3.Ev3Communicator - token QDR25DDG is approved by a user.
06:47:34.329 [qtp1433292785-15] INFO d.f.i.r.r.ev3.Ev3CommunicationData - Robot request for token approval terminated SUCCESSFULLY. Time elapsed: 0
06:47:34.329 [qtp1433292785-35] INFO d.f.i.r.j.r.all.ClientAdmin - success: token QDR25DDG is registered in the session
06:47:36.954 [qtp1433292785-34] INFO d.fhg.iais.roberta.util.ClientLogger - log entry: [[TIME]] 2 msec: set token 'QDR25DDG'
[... cut ...]
	at com.sun.jersey.server.impl.model.method.dispatch.AbstractResourceMethodDispatchProvider$ResponseOutInvoker._dispatch(AbstractResourceMethodDispatchProvider.java:205) [jersey-server-1.18.2.jar:1.18.2]
	at com.sun.jersey.server.impl.model.method.dispatch.ResourceJavaMethodDispatcher.dispatch(ResourceJavaMethodDispatcher.java:75) [jersey-server-1.18.2.jar:1.18.2]
	at com.sun.jersey.server.impl.uri.rules.HttpMethodRule.accept(HttpMethodRule.java:302) [jersey-server-1.18.2.jar:1.18.2]
	at com.sun.jersey.server.impl.uri.rules.ResourceClassRule.accept(ResourceClassRule.java:108) [jersey-server-1.18.2.jar:1.18.2]
	at com.sun.jersey.server.impl.uri.rules.RightHandPathRul

ensonic, 2016-04-13:
Reusing the key was meant as a convenience for developers. I reverted that part. I'll see if I can make the UI nicer to wait for the disconnect to happen before allowing to re-connect - https://github.com/OpenRoberta/robertalab-ev3dev/issues/18

ensonic, 2016-04-13:
I still think we shoudl allow robots to reuse the key, since we'll have this behavious for robots with static keys.

closed 2016-04-13 by ensonic