OpenRoberta/robertalab #72: Executing program on ev3dev causes display issue

bjost2s, 2016-03-02:
there is a strange collection of small letters/symbols on the ev3 display during executing this program
[image]

bjost2s, 2016-03-02:
Maybe caused also by a wrong ev3dev version?

dlech, 2016-03-02:
I don't think the ev3dev version puts the console into graphics mode or raw keyboard mode. The symbols you are seeing are probably escape codes from when keys were pressed in a previous program.

I have a python program with a proper-ish way to do this - well switching to graphics mode anyway. <https://github.com/dlech/ev3dev-photo-booth/blob/master/ev3dev-photo-booth.py>. The framebuffer and virtual terminal code should be made into separate libraries and integrated with ev3dev-lang-python.

dlech, 2016-03-02:
I've actually started work on cleaning up the code for separate libraries. I haven't published it anywhere yet, so if anyone would like to take over working on it, let me know.

ensonic, 2016-03-02:
I am turning off the cursor, but there seems to be indeed still some issue. I'll take a look next week.

dlech, 2016-03-02:
Even if you turn off the cursor, you still have stdin echoing and stdout/stderr printing on the console. Any time you press a button, you will see things like `^[` printed on the screen.

bjost2s, 2016-04-06:
Is this still an issue?

ensonic, 2016-04-07:
Yes, I did not had time to work on this, but I'll move it to robertalab-ev3dev, since it is unrelated to the server.

closed 2016-04-07 by ensonic

ensonic, 2016-06-03:
I just fixed it. Will release a new version of the package soon.