OpenRoberta/robertalab #76: index.html und require.js

rbudde, 2016-03-04:
the main entry file `index.html` contains the lines
```
<script type="text/javascript" src="js/app/simulation/robertaLogic/constants.js"></script>
<script type="text/javascript" src="js/app/simulation/robertaLogic/program.builder.js"></script>
<script type="text/javascript" src="js/libs/code-prettify/prettify.js"></script>

<script data-main="js/main" src="js/require.js"></script>
```
at least the first two scripts (we own) should be loaded with require. The last one is code supplied by others, but loading with require would be much more robust and safe.
Btw: it would nice to cleanup `index.html`.

cvejoski, 2016-03-04:
When we put in require there are some problems with loading the prettify. Next week I will work more on cleaning the client side code and loading all js files with require

ensonic, 2016-03-06:
I'll change the pretty.js part to use require.

bjost2s, 2016-04-06:
I think this is fixed, please confirm and close, if so

ensonic, 2016-04-07:
two files still need to be converted:
https://github.com/OpenRoberta/robertalab/blob/develop/OpenRobertaServer/staticResources/index.html#L28_L30

bjost2s, 2016-04-18:
we discussed to remove pace.js and convert the functionality to pure css

bjost2s, 2016-04-18:
pace.js and dependencies are removed, needs some testing on safari etc

closed 2016-04-25 by bjost2s