OpenRoberta/robertalab #86: repeat until not working in the simulation

ensonic, 2016-03-10:
```
16:50:39.223 [qtp2134196695-14] ERROR d.f.i.r.j.r.all.ClientProgram - Exception. Error ticket: E-4
de.fhg.iais.roberta.util.dbc.DbcException: Invalid repeat mode
	at de.fhg.iais.roberta.syntax.codegen.ev3.Ast2Ev3JavaScriptVisitor.appendRepeatStmtCondition(Ast2Ev3JavaScriptVisitor.java:849) ~[OpenRobertaRobot-1.3.2-SNAPSHOT.jar:na]
	at de.fhg.iais.roberta.syntax.codegen.ev3.Ast2Ev3JavaScriptVisitor.visitRepeatStmt(Ast2Ev3JavaScriptVisitor.java:306) ~[OpenRobertaRobot-1.3.2-SNAPSHOT.jar:na]
	at de.fhg.iais.roberta.syntax.codegen.ev3.Ast2Ev3JavaScriptVisitor.visitRepeatStmt(Ast2Ev3JavaScriptVisitor.java:100) ~[OpenRobertaRobot-1.3.2-SNAPSHOT.jar:na]
	at de.fhg.iais.roberta.syntax.stmt.RepeatStmt.accept(RepeatStmt.java:101) ~[OpenRobertaRobot-1.3.2-SNAPSHOT.jar:na]
	at de.fhg.iais.roberta.syntax.Phrase.visit(Phrase.java:113) ~[OpenRobertaRobot-1.3.2-SNAPSHOT.jar:na]
	at de.fhg.iais.roberta.syntax.codegen.ev3.Ast2Ev3JavaScriptVisitor.generateCodeFromPhrases(Ast2Ev3JavaScriptVisitor.java:767) ~[OpenRobertaRobot-1.3.2-SNAPSHOT.jar:na]
	at de.fhg.iais.roberta.syntax.codegen.ev3.Ast2Ev3JavaScriptVisitor.generate(Ast2Ev3JavaScriptVisitor.java:114) ~[OpenRobertaRobot-1.3.2-SNAPSHOT.jar:na]
	at de.fhg.iais.roberta.javaServer.restServices.all.ClientProgram.command(ClientProgram.java:226) ~[OpenRobertaServer.jar:na]
```

ensonic, 2016-03-10:
I'll take a look.

bjost2s, 2016-03-10:
All 'exports' blocks have to be implemented for the simulation. Please use or link to #28 / #29 which are still in progress

cvejoski, 2016-03-23:
This is fixed with commit  https://github.com/OpenRoberta/robertalab/commit/50f677aa274d67e47e97efdf1ff27b17b0b77663.

closed 2016-03-23 by cvejoski