OpenRoberta/robertalab #89: Customizable simulation map

alesolano, 2016-03-11:
I don't know if we can change the OR simulation map at our choice. It would be marvelous. 

Thank you!

bjost2s, 2016-03-13:
Yes indeed. We should keep this in mind for one of the next releases.

ensonic, 2016-03-14:
For this we need to find a way to store assets (images and sounds) with a program.

closed 2017-02-04 by bjost2s

ensonic, 2017-02-05:
@bjost2s where is the new simulation background stored?

bjost2s, 2017-02-06:
No, not yet. It is stored in the browsers' storage until we are clear about what we would like to store with the project.
Users may use the custom background for different projects, so they just have to upload it once, currently.

alesolano, 2017-02-20:
Will it be possible to include obstacles? I mean, if I upload a background with walls drawn on it, I want them to be solid enough to activate the touch sensor.

bjost2s, 2017-02-27:
This is not possible with a simple upload of the image, because it is impossible to extract information about "walls" from there. This may be possible once we support custom robots and environments, but there is no plan for it. The reason is that the Open Roberta Project is there to support learning with hands on real robots. The simulation should not substitude the real robot, it should just give an idea how it works.

alesolano, 2017-02-27:
Imagine a classroom with 20 students and just one robot. It would be fantastic if every student could work at home, testing the software he/she just developed in the simulator and then, at school, download it into the robot.

bjost2s, 2017-02-28:
we are thinking about implementing a maze, would that help?