OpenRoberta/robertalab #90: Need a way to store assets with a program

ensonic, 2016-03-14:
robertalab#89 asks for custom simulation maps. Also for the showPicture() and playFile() blocks, it would be nice to be able to have custom assets.

In the java case we could have them in the jar, but that would make the uploaded slower - ideally we only send them as needed. So for java we could have a prog.jar and prog-data.jar.

In python we'd include them as a module. The module can be built-in or a separate file.

If we go for separate files, we can upload them as needed.

ilovefroyos, 2018-03-20:
Hi @ensonic :) ! I'm an undergraduate student at The University of British Columbia studying computer science, and I'm interested in contributing to Open Roberta Lab for GSoC. Has this issue already been claimed/assigned/addressed for/by any other applicants?

rbudde, 2019-06-26:
GSOC project

closed 2019-06-26 by rbudde