OpenRoberta/robertalab #108: include pictures with the code and remove from libs

ensonic, 2016-04-06:
Right now we have a built-in list of pictures for the showPicture block. Those are compiled into the client libraries. Extending them would require library updates and would also increase the library size.

Since a picture is only 178*128/8 = 2848 bytes. I propose to store them on the server and include them in the generated code (as bytearray). This way we can add more picture with new releases and don't have to update the client libraries. This would also be a first step to allow custom pictures.

cvejoski, 2017-01-03:
ev3 implementation: f7a3d1f

cvejoski, 2017-01-03:
26f04bc

cvejoski, 2017-01-03:
ev3 implementation: f7a3d1f,	26f04bc

ensonic, 2017-01-03:
I'll do the changes for ev3dev tomorrow. One more thought still:
What about adding the used images to the brick-config and then referencing it from there? This would make the generated code more readable, since we would not have the long image data string in the the hal.drawPicture() call.

bjost2s, 2017-02-01:
What is the current state?

ensonic, 2017-02-02:
Its is working. The code is not perfect since we now include the image 'string' directly in the call. We could make a separate ticket for having a global asset storage (dictionary, hashtable).

closed 2017-10-02 by bjost2s