OpenRoberta/robertalab #111: api urls changed

ensonic, 2016-04-13:
we switched to a new scheme with a "/rest" segment in the url for pushcmd and download. On the python side I just tried without /rest and upon failure switched to using /rest. Now this changed and the pushcmd works without rest, but download still needs it. I am now refactoring my code to always try both until we release 1.4. But please consider reverting this change.

ensonic, 2016-04-13:
I will push a new ev3dev update that is more flexibe (fix it in git already).

bjost2s, 2016-04-25:
I think it is done for 1.4.0

ensonic, 2016-04-26:
I've fixed this with the robertalab-ev3dev-1.4.0 release which is going to theses days. But on the server side this was not fixed and I am not sure which commit changed it.

rbudde, 2016-04-26:
After having seen many server errors due to octet-stream-content, I've committed changes in develop [1edde4f](https://github.com/OpenRoberta/robertalab/commit/1edde4fb7d5804665fd7b8be0fef13786abb9cbd) to allow both download and update to use the deprecated api again. Obviously we missed to test some parts of the upgrade process. This change must be tested with an EV3/lejos 1.3 and then deployed.

BeateJ, 2016-04-26:
It is fixed for the public lab and tested. It works fine! All users could update their robots yesterday

closed 2017-07-21 by cvejoski