OpenRoberta/robertalab #116: OR Lab on win exploerer

bjost2s, 2016-04-14:
Open Roberta Lab is not running on windows explorer (thorsten)

bjost2s, 2016-04-14:
seems to be an jquery/require issue?

cvejoski, 2016-04-15:
The problem is with web audio api, it is not supported by IE 11 (no surprise). I have not seen any solution yet.

ensonic, 2016-04-15:
Can we just wrap all audio calls and make them do nothing on IE? Also Edge supports it, for those that would like to stick with MS.

cvejoski, 2016-04-15:
Yes, I am working on that, they will be just ignored.

rbudde, 2016-04-22:
there are a lot of errors:
* IE < 11 will never work, as "const" and "WeakMap" are not supported
* In IE11: blockly is compressed with "strict". File blockly_compressed.js line 995 (definition of function Blockly.RobControls.prototype.createButton_) contains two "fill" properties. Other browsers than ie seem to ignore this error. After removing one "fill" the lab starts in ie 11.

@bjost2s **Beate: please correct the error!**
* simulation works, but ...
* editing the content of blocks (e.g. the "0" of a number block) does not work
* writing text to the ev3 screen doesn't work

I suppose, there are much more errors. I'll try to find some time to test and add further comments to this ticket.

rbudde, 2016-04-22:
in ie11
* the welcome screen has an ugly appearance
* the "center" command in the zoom button group doesn't work properly (looks like a wrong window size is used for centering). Sometimes nothing happens, sometimes the wrong position is calculated
* starting the simulation sometimes hides (parts of) the program, after terminating simulation the program is outside the window, using zoom/center lets the program re-appear. This may be effected by using zoom before. It's more likely, that window sizes are wrong here, too.

bjost2s, 2016-04-25:
more work for 1.5

closed 2017-01-27 by bjost2s