OpenRoberta/robertalab #123: French Translation

Wodric, 2016-04-29:
Hi all,

On source code I found partial french translation. It could be interesting to finish it's integration. 

What do you think about this?

bjost2s, 2016-04-29:
In general we use the translations for the blocks from blockly. All non blockly blocks (robot blocks) and meny stuff will be translated in a google table. Someone else just started with the french translation. Would you like to help? If so I would send you an invitation to the doc ;-) 
Many thanks for you interest in Open Roberta

Wodric, 2016-05-02:
You can invite me, I don't know when I will have time to contribute, I will help as much as possible.

ensonic, 2016-06-06:
Please contribute through the doc.

closed 2016-06-06 by ensonic