OpenRoberta/robertalab #124: SSO connector

Wodric, 2016-04-29:
Hi,

In french education system, there are lot of educationnal sofwares which are connect by SSO (mostly with shibboleth or CAS). The integration Open Roberta  in these could be much more easier. 
I didn't see anything about it on Open Roberta. What do you think about the idea? 

 I could get involved in this.

bjost2s, 2016-04-29:
This sounds interesting! A long time ago we have discussed this, but unfortunately nobody of us has had time for this. Currently the Open Roberta Team is focussing on integrating new robot systems, but it would be great if you would like to help us. Let's discuss necessary steps here.

ensonic, 2016-04-30:
A first step would be to collect which SSO solutions are used in various countries (with links to who provides them - for 'CAS' I find a lot of hit, but nothing that looks like SSO). Ideally we can figure on what technology they are based (e.g. OAuth2). This could be done on a wiki page liked from here.

Wodric, 2016-05-02:
I let you create the wiki page, can't contribute to it.
As I know, CAS and Shibboleth is very used in France, I already heard about OAuth2 in some systems. 

LMS (Moodle for exemple) are using LTI protocol to import courses or exercise.

rbudde, 2019-06-26:
not feasable at the moment

closed 2019-06-26 by rbudde