OpenRoberta/robertalab #128: NXC code generation

bjost2s, 2016-05-13:
This is the first step to integrate the NXT into the Open Roberta Lab. For the implementation of the code generation we will use the branch NXT_code_generation. After that the OpenRobertaRobot project will become a common robot project and the ev3 and nxt code generation implementations will move to seperate projects /repos.

The generated code will be NXC code which can afterwords be compiled to bytecode for NXT.

ensonic, 2016-07-22:
@elhele regarding commit 511679c, I was considering to rename the UsedSensorVisitor into a UsedResourceVisitor and make it track sensors, actors. This could also track the timers. What do you think? Otherwise consider renaming TimeVisit to UsedTimerVisitor.

elhele, 2016-07-22:
@ensonic It was a visitor for NXT where we did't have the "UsedSensorVisitor". But I could name it "UsedResourceVisitor", it would make sense in case I'd be adding anything else there.

ensonic, 2016-07-22:
@elhele if it is NXT only, then just rename it to UsedTimerVisitor to be more precise. I was about to rename the UsedSensorVisitor to UsedResourceVisitor since I am going to expand it. If your visitor is goign to track other resources as well, you can use the same name for consistency since they'll be in different packages.

elhele, 2016-07-22:
@ensonic ok, I'll rename it so.

closed 2016-09-21 by cvejoski