OpenRoberta/robertalab #130: NXT compiler workflow

bjost2s, 2016-05-13:
TODO

BeateJ, 2016-06-13:
will be done in #131

dpyka, 2016-07-28:
NXT roundtrip is now working for Windows on server side and Windows on client side.

dpyka, 2016-07-28:
Some paths are resolved quite strange in the NxtCompilerWorkflow, but they are working somehow. Taken from the output of the compiler binary:
# Status: Current file = "E:\Development\Repositories\robertalab\**OpenRobertaServer\../OpenRobertaServer**/userProjects/YSW7S26A/NEPOprog.nxc"
# Status: Include path = .\;E:\Development\Repositories\robertalab\OpenRobertaServer\;E:\Development\Repositories\robertalab\RobotNXT\resources\;resources**//hal.h\**
# Status: Searching for file: E:\Development\Repositories\robertalab\OpenRobertaServer**\.\hal.h**
# Status: Include path = .\;E:\Development\Repositories\robertalab\OpenRobertaServer\;E:\Development\Repositories\robertalab\RobotNXT\resources\;resources**//hal.h\**
# Status: Current file = "E:\Development\Repositories\robertalab\**OpenRobertaServer\../OpenRobertaServer**/userProjects/YSW7S26A/NEPOprog.nxc"
# Status: Current file = "E:\Development\Repositories\robertalab\**OpenRobertaServer\../OpenRobertaServer**/userProjects/YSW7S26A/NEPOprog.nxc"
# Status: Include path = .\;E:\Development\Repositories\robertalab\OpenRobertaServer\;E:\Development\Repositories\robertalab\RobotNXT\resources\;resources**//hal.h\**

closed 2016-09-21 by cvejoski