OpenRoberta/robertalab #132: Running in Docker

psychemedia, 2016-05-21:
I've successfully built *robertlab* in a Docker container [[Dockerfile](https://gist.github.com/psychemedia/6f5b6553669eb28536a8834ec2490469)] but notice a comment that the default `ip=0.0.0.0` setting is *[ok almost everywhere, but __not with docker__](https://github.com/OpenRoberta/robertalab/search?q=%22not%20with%20docker%22&type=Code&utf8=%E2%9C%93)*.

Is there any guidance on how this should be set for running successfully from a container?

closed 2016-05-21 by psychemedia

psychemedia, 2016-05-23:
Ah.. I see what the original issue was: I *can* connect to Roberta running in a container from localhost, but I can't connect to it from elsewhere on the localhost network *unless* I set up a port forwarding rule in Virtualbox, as described (here](https://blog.ouseful.info/2016/05/22/exposing-services-running-in-a-docker-container-running-in-virtualbox-to-other-computers-on-a-local-network/)

WizGeek, 2017-03-16:
For Roberta in a Docker container, what is needed to enable WiFi access to the EV3 brick?  
Details:
- Windows 10 Pro 64-bit:  Version 10.0.14393 Build 14393.
- Chrome browser:  Version 56.0.2924.87.
- Internet Roberta:  Connects wireless via 8-character token successfully.
- Docker Roberta:  Browser accessible via localhost:1999.
- PROBLEM:  Unable to connect via WIRELESS to the EV3 brick--it times out without finding the token.

What is needed for Docker or Hyper-V to allow access via WiFI/wireless?