OpenRoberta/robertalab #146: Modifying Bluetooth blocks

elhele, 2016-06-17:
- Block "connect to robot name" is not needed for nxc. 
- Check of connection is implemented in a hal.h function, so calling of this function could replace "wait for connection block".
- Block for sending messages requires the following input: output variable (accepts string, boolean or number), connection (int, 1-3 for master, 0 for slaves), outbox address (int).
- Block for receiving messages requires the following input: inbox address (int).

It is also possible to implement the following functionality: RemoteStartProgram, RemoteStopProgram, RemoteKeepAlive, RemotePlayTone and some other. However, I'm not entirely sure that they are needed now.

ensonic, 2016-06-18:
I vote for reviewing these blocks too. Whether they use bluetooth or something else should be transparent. For robots with wifi, we could e.g. send the messages via the server.

bjost2s, 2017-02-25:
fixed in OpenRoberta/blockly@2cd1519e3f1076fdc4cd02de3b19df59ce202e47

closed 2017-02-25 by bjost2s