OpenRoberta/robertalab #150: NXT new / modified blocks and toolboxex for configuration and programming

bjost2s, 2016-06-24:


bjost2s, 2016-06-25:
the following blocks can't be supported, so they are not in the toolbox anymore, see #147:
- robActions_display_picture 
- robActions_play_file
- robActions_brickLight_on
- robSensors_gyro_reset
- robSensors_gyro_getSample
- robText_join
- robLists_repeat
- robLists_indexOf
- robLists_getSublist
- robCommunication_startConnection
- robCommunication_waitForConnection

the following blocks are new:
- robActions_sensorLight_on
- robSensors_sound_getSample
- robSensors_light_getSample - two modes, light and ambient light
- robCommunication_connection
 
the following blocks are modified:
- all blocks with motor ports, removed port D
- button block, remove up, down
- ultrasonic sensor, remove mode presence
- send and receive message:

 `<block type="robCommunication_receiveBlock">`
     ` <mutation data_type="Number"></mutation>`
      `<field name="TYPE">Number</field>`
      `<field name="PROTOCOL">BLUETOOTH</field>`
      `<field name="CHANNEL">0</field>`
      `<data>nxt</data>`
   ` </block>`

bjost2s, 2016-06-25:
There is still one new block missing: that the user can check for a bluetooth connection

elhele, 2016-06-26:
Hello. What is wrong with robActions_play_file? I thought there was a function for it: "PlayFile("fileName.rso");" http://bricxcc.sourceforge.net/nbc/nxcdoc/nxcapi/ex__play_file_8nxc-example.html#a0

bjost2s, 2016-06-26:
The problem is, that there are no files on the robot to display. The user should have had uploaded these files manually before the program is executed. Our communication protocol currently does not allow uploading files.

elhele, 2016-06-26:
Ok, thank you.

bjost2s, 2016-06-27:
To see and use the new / modified blocks, please have a look at the nxt_code_generation branch. I had to remove the new sensors from the standart configuration, not to block the whole code generation. Just take them out of the toolbox to implement them on serverside.

elhele, 2016-06-28:
The button "exit" would work correctly only with enhanced firmware, so we need to remove it from brick sensor button list.

bjost2s, 2016-07-21:
TODO remove mode presence from ultrasonic sensor

closed 2016-09-21 by bjost2s