OpenRoberta/robertalab #156: Infinite loop of error dialogs

drigz, 2016-07-14:
I got this when changing wifi network while the lab was loading.

[image]

Clicking OK in the dialog makes it disappear and then immediately reappear.

drigz, 2016-07-14:
This appears to be intentional: [handleServerErrors()](https://github.com/OpenRoberta/robertalab/blob/605a0dcfd151fac3b4b89285b2e7ea01b6fa4f5d/OpenRobertaServer/staticResources/js/app/roberta/roberta.js#L175) installs an event handler that reshows the message when it is hidden.

If there's no point in allowing the user to use the application without server access, we should hide the OK button. Would that be better?

ensonic, 2016-07-14:
+1 for hiding the OK button. Also regarding the typo, imho you want to fix it in en.json.

bjost2s, 2016-07-14:
+1
if the internet connection is bad people claimed that they ran into this issue very often. Maybe we could could think about reconnecting automatically?

drigz, 2016-07-15:
I definitely think reconnecting automatically would be best. I discussed this briefly with Thorsten:

* if we get the error before requirejs emits the ready event, we can starting pinging and reload the page when the pings work
* if we get the error after requirejs emits the ready event, we shouldn't need to reload, so we should keep the dialog up until the pings work

larsrc-google, 2016-07-15:
I get this dialog when running against a local server on a laptop and closing the laptop for a while then coming back to it.