OpenRoberta/robertalab #163: Store drafts in browser

larsrc-google, 2016-07-14:
To prevent losing programs, especially on mobile devices that can readily lose the tab, unsaved drafts should be stored in the browser and reloaded on startup.

bjost2s, 2016-07-14:
Do we need to inform the user about these storing something in the browser? Is it handled similar like cookies? If so we should think about

ensonic, 2016-07-18:
https://github.com/OpenRoberta/robertalab/pull/171

larsrc-google, 2016-07-18:
As far as I can tell, yes. Probably just need to update the existing cookie
warning to also mention local storage.

-Lars

bjost2s, 2016-07-18:
As we are providing software for kids, we should think about this in detail, what if people don't allow using local storage?

larsrc-google, 2016-07-18:
We should check if access attempts throw exceptions or just return nothing.
If the latter, we're fine.

-Lars