OpenRoberta/robertalab #177: Ambient mode

elhele, 2016-07-15:
I've tested SENSOR_TYPE_COLORNONE, the ambient light mode, on NXT with light and color sensors.
With light sensor it just turns on very bright red light and  can only detect this light which makes it not really useful mode.
However, with color sensor it actually turns off the light, returns sensible values and acts like a real ambient mode.
So, I guess, for NXT it should also be in color sensor block.
Light sensor should be left without any modes and color sensor should have simple "COLORFULL" and ambient "COLORNONE" modes.

bjost2s, 2016-07-15:
The nxt light sensor has no internal modes it just measures the reflected light. If our user would like to use the sensor in light mode, turn the light on. The returned values are between 0 and 100. To test this use a black line on a white background and hold the sensor 0,5 cm above.
The ambient light mode is  simply measuring the light without the internal led turned on.

bjost2s, 2016-07-15:
The nxt color sensor works similar to the ev3 color sensor, I think there is one mode less. This is not a standard sensor for nxt so it is an extra sensor for the nxt, not in the standard configuration nor in the beginner toolbox

elhele, 2016-07-15:
Ok, so we will leave the modes for color sensor as well? I just remember that we were going to remove them.

bjost2s, 2016-07-15:
No! We have to check if the colorsensor from NXT could support the same modes than on ev3
Light
ambient light
color are supported
rgb?

bjost2s, 2016-07-15:
#150

elhele, 2016-07-15:
I've tested. It can support color, light and ambient light (I've already implemented it and can commit soon).
There is no RGB mode, only red green and blue separately.

bjost2s, 2016-07-15:
please use the red light for light (red) mode, this colors gives best results

bjost2s, 2016-07-21:
TODO remove RGB mode from blockly block

closed 2016-09-21 by bjost2s