OpenRoberta/robertalab #189: Can't connect with OR Lab because of non-uptaded firmware

alesolano, 2016-08-02:
Hello!
After four months of inactivity, I have come back to EV3 and OR Lab, but it turns out that I can't connect my EV3 with OR.
I still use the last release version of [ev3dev](https://github.com/ev3dev/ev3dev/releases/tag/evb-ev3dev-jessie-2016-01-25), from January. 
Four months ago, it worked perfectly. 

This is what I tried:

[image]

And when I run it:
[image]

And if I select Update now:

[image]

And if I disconnect the EV3 and connect again (with the new pairing code) nothing new happens and it ask me again to update my robot.

Thank you!

cvejoski, 2016-08-03:
This bug finally got fixed in the development version that we soon publish as 2.0. In any case the "openrobertalab" on the ev3dev needs to be updated.

For that you need to ssh into the running ev3. From linux this would be
ssh robot@ev3dev.local
From windows putty can be used. The password is 'maker' as documented on ev3dev.org

A normal update would be to run
sudo apt-get update
sudo apt-get upgrade

This takes ages and also right now would update the kernel, which is not recommended since the driver apis changed.

This ticket https://github.com/OpenRoberta/robertalab-ev3dev/issues/19 describes a faster way. If you don't mind to wait a bit (lunch time), just run:
sudo apt-get update
sudo apt-get install --only-upgrade openrobertalab

We're discussing with the ev3dev folks how to add some package updating the the brickmnan ui.

alesolano, 2016-10-03:
What should I do now that we have a new release?

Would it be OK if I use the update and install?
```
sudo apt-get update
sudo apt-get install --only-upgrade openrobertalab
```

ensonic, 2016-10-03:
Your issue was a server side problem that is fixed in the new release that soon will be online (2.0).
Still making sure the openrobertalab package on the ev3 is up-to-date is a good idea too, I just released a new version last week.

alesolano, 2016-10-04:
Solved!
The commands worked.
Thank you.

closed 2016-10-04 by alesolano