OpenRoberta/robertalab #190: As a user I'd like to quickly reconnect my robot if the connection was lost

ensonic, 2016-08-03:
It would be nice for the robots to only regenerate a new token on collisions or when they reboot. It the token entry dialog in the UI keeps the last used token (while the browser is open), all the user needs to to is to press connect on the robot and enter on the browser-ui.

bjost2s, 2016-08-04:
As far as I understand, this issue needs an update for the EV3lejos menu, because the EV3lejos is generating all the time a new token. We have to change that first, otherwise the user has to delete the token in the form all the time.
Also we have to change the usb program and the chrome app?
Currently there is no update planned for the ev3lejos.
Should we postpone this to 2.0?

ensonic, 2016-08-04:
I think all we need to make sure is that the text is selected. This way one can click okay to reuse it or type to overwrite. I had received this as a feature request on the robertalab-ev3dev side. Since the change is simple I would prefer to not to postpone this. Also initially the token is empty and re-connecting should be the exception.

rbudde, 2019-06-26:
solved

closed 2019-06-26 by rbudde